xMasterX/all-the-plugins #6: [Flipper_Catalog] apps_data files are missing for Barcode_app

theblackhole, 2023-07-18:
Hello,

I don't know if the issue is on your side or if it is an issue on the catalog side so I will create an issue for both.

Linked issue:

TLDR: it seems like required app_data files are missing when installing apps from the Hub/Catalog 

Details :
I have an issue with the [barcode_app](https://lab.flipper.net/apps/barcode_app) installed with the android app via the Hub/Catalog: when I try to load a Code 128 barcode, it says "Missing encoding table".
After reading the [installation instructions on the original github repository](https://github.com/Kingal1337/flipper-barcode-generator#installing) I found that it needs enconding table files in a `apps_data/barcode_data` folder.
This folder is missing on my Flipper Zero:
[image]
The issue disappeared after I added the missing folder and files manually:
[image]

xMasterX, 2023-07-18:
Files are not packed into app itself, and cannot be shipped separately, catalog doesnt allow to add external assets 
I will remake this app to have them all packed into this app, a bit later

RogueMaster, 2023-07-18:
Or better yet, Flipper Devices gives a way to pack files for the SD with the app install...

xMasterX, 2023-07-18:
> Or better yet, Flipper Devices gives a way to pack files for the SD with the app install...

This is exactly what i said

xMasterX, 2023-07-19:
Fixed, will be updated in catalog soon

closed 2023-07-19 by xMasterX