xMasterX/all-the-plugins #13: Spectrum Analyzer Credit

ALEEF02, 2023-09-07:
Hey,

I've done work on the Spectrum Analyzer and had proper credit until the app move-over. Please add my username to the credits. I can also do it myself and make a PR, but I'm sure you guys have a specific way you want it done.

https://github.com/DarkFlippers/unleashed-firmware/pull/557
https://github.com/DarkFlippers/unleashed-firmware/pull/553
https://github.com/DarkFlippers/unleashed-firmware/pull/550

xMasterX, 2023-09-07:
Hi
It has your name in app authors here:
base_pack/spectrum_analyzer/application.fam

Also i updated the readme now:
https://github.com/xMasterX/all-the-plugins/commit/1c79585c989ceae7f18ec9eef772af86038b421a

closed 2023-09-07 by xMasterX

ALEEF02, 2023-09-07:
Looks great, thank you.