xMasterX/all-the-plugins #16: universal remote: missing elements tv

nanderer, 2023-09-18:
Hi,

Its missing my cheap Chinese tv branded with elements. Maybe you could include it:
https://flpr.app/sf#path=infrared/ElementsTV.ir&key=iEzearNdxWz-nKal596Iww&id=VVbh7E

Thanks

nanderer, 2023-09-18:
Why is there no no upload button in the mobile version @GitHub?

And now i have to zip it to ;(

[ElementsTV.ir.zip](https://github.com/xMasterX/all-the-plugins/files/12653506/ElementsTV.ir.zip)

There you (finally) go. Consider switching to a real open source git platform (like bitbucletorgitlab)

xMasterX, 2023-09-18:
> There you (finally) go. Consider switching to a real open source git platform (like bitbucletorgitlab)

no

xMasterX, 2023-09-18:
Create issue in correct repo

closed 2023-09-18 by xMasterX

nanderer, 2023-09-19:
Plese tell me where