xMasterX/all-the-plugins #29: Possible license/copyright violations

q3w3e3, 2023-11-18:
A list of application reproduced in this repo that have no license and thus are [cannot be reproduced, distributed or have derivative works created from](https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/licensing-a-repository#choosing-the-right-license)

https://github.com/panki27/Metronome
https://github.com/wh00hw/MorseCodeFAP
https://github.com/Nem0oo/flipper-zero-bluetooth-trigger
https://github.com/leo-need-more-coffee/flipperzero-4inrow
https://github.com/nmrr/flipperzero-pong
https://github.com/Daniel-dev-s/flipperzero-slots
https://github.com/Willzvul/Snake_2.0
https://github.com/RouNNdeL/flipper-tictactoe-multi
https://github.com/nmrr/flipperzero-geigercounter
https://github.com/aureli1c/flipperzero_GPIO_read
https://github.com/spin7ion/flipper-servotester
https://github.com/Nitepone/flipper-intervalometer
https://github.com/invalidna-me/flipperzero-ocarina
https://github.com/Round-Pi/flipperzero-text2sam
https://github.com/DrZlo13/flipper-zero-music-tracker
https://github.com/spheeere98/mifare_fuzzer
https://github.com/scrolltex/flipper_analog_clock
https://github.com/nymda/FlipperZeroBrainfuck
https://github.com/0w0mewo/fpz_cntdown_timer
https://github.com/Krulknul/dolphin-counter
https://github.com/dunaevai135/flipper-zero-hex_editor

xMasterX, 2023-11-18:
All those applications were derived from a GPLv3 project, and therefore are all covered by GPLv3
You are throwing false claims and also you prefer to ignore MY work that I did to many of those plugins
I am considered as co-author for most of them since I made my modifications
Also I know most of the authors of those plugins, and they know me, no one of them contacted me to remove their app
If anybody of them wants me to remove their app, they can contact me in any possible way, via discord, telegram, or in github
But looks like its your problem, not theirs.

Apps made with flipper firmware SDK is licensed under GPLv3 as they was made inside GPLv3 project, cannot work independently, and require flipper firmware SDK to be able to even compile
There is no apps that wasn't made for flipper's firmware in this repo

closed 2023-11-18 by xMasterX

q3w3e3, 2023-11-18:
hi, sorry I did not mean to imply I don't appreciate the work you do or anything of the sort. or that I want these removed.

if all apps here are gol, they must be distributed with a  copy of gpl, some even explicitly have a bad license... playing fast and loose with licensing is something I do all the time too.

I wish it wasn't the case but this is still an issue Downstream of you when these apps are in the catalog and are distributed on the app store.

I bring up this concern not because I want the apps deleted.

but because I want to see them continue to be accessible, maintained and distributed, but without proper licensing that could become an issue.

again, I appreciate the work you do, I don't want to see the flipper app get DMCAd or similar, and I was opening these issues to try to bring awareness to some procedural issues that could lead to a mess in the future.