xMasterX/all-the-plugins #37: tetris_game 7-bag randomizer not being reset after a game over

Alloy-Lou, 2024-01-03:
The 7 bag randomizer is not being reset after a game over or when starting a new game

How to reproduce : 
- Drop exactly 5 piece
- Press back button ("The game is now paused")
- Press menu button (Start a new game)
- Notice the first dropping piece doesn't follow the 7-bag rule, with a bit of luck you'll notice the falling piece and the next piece being the exact same.

Also works after a game over.
Not sure that when pressing the menu button while the game is paused, starting a new game is an intended behavior. 

I also have a few suggestions to improve the game (such as implementing counter-clockwise rotation with up button, or add hold feature, improve piece preview with more previews, add lock delay, improve scoring system with b2bs... -> being closer to the official Tetris Guideline), I'd happily contribute.

noiob, 2024-02-06:
the center button isn't supposed to start a new game when paused, that's a bug

Alloy-Lou, 2024-02-10:
I fixed these bugs and I'm currently working on additional features, feel free to dm me on Discord (@alloy_lou), maybe we could work together?

noiob, 2024-02-10:
sorry, I pretty much just fixed some stuff that annoyed me in the game so it'd be more fun for me. I'm not really a C dev :D

Alloy-Lou, 2024-02-11:
oh i see, i recently started C too. Would you at least be interested in testing the game so I could have feedback, or find bugs i wouldn't have seen alone?