xMasterX/all-the-plugins #103: Bug on asteroids

Cyshots, 2024-04-28:
Hey i see long lines and its goes very slow at times which i cant control my jet

karlellis, 2025-02-01:
Hi,
 I'm not so skilled so I used copilot.
To avoid drawing segments across the screen when a polygon crosses the boundary, you can split the line segments into smaller segments that stay within screen bounds.
Changing the function "draw_poly" at line 276 of "flipper-asteroids/app.c"
```C
void draw_poly(Canvas* const canvas, Poly* poly, uint8_t x, uint8_t y, float a) {
    Poly rot;
    rotate_poly(&rot, poly, a);
    canvas_set_color(canvas, ColorBlack);
    for(uint32_t j = 0; j < rot.points; j++) {
        uint32_t a = j;
        uint32_t b = j + 1;
        if(b == rot.points) b = 0;
        canvas_draw_line(canvas, x + rot.x[a], y + rot.y[a], x + rot.x[b], y + rot.y[b]);
    }
}
```
with this one:
```C
void draw_poly(Canvas* const canvas, Poly* poly, uint8_t x, uint8_t y, float angle) {
    Poly rotated;
    rotate_poly(&rotated, poly, angle);
    canvas_set_color(canvas, ColorBlack);

    for(uint32_t i = 0; i < rotated.points; i++) {
        uint32_t next = (i + 1) % rotated.points;

        int x1 = (int)(x + rotated.x[i]) % SCREEN_XRES;
        int y1 = (int)(y + rotated.y[i]) % SCREEN_YRES;
        int x2 = (int)(x + rotated.x[next]) % SCREEN_XRES;
        int y2 = (int)(y + rotated.y[next]) % SCREEN_YRES;

        if (x1 < 0) x1 += SCREEN_XRES;
        if (y1 < 0) y1 += SCREEN_YRES;
        if (x2 < 0) x2 += SCREEN_XRES;
        if (y2 < 0) y2 += SCREEN_YRES;

        // If the line crosses the screen boundary
        if (abs(x1 - x2) > SCREEN_XRES / 2) {
            if (x1 > x2) {
                canvas_draw_line(canvas, x1, y1, SCREEN_XRES, y1);
                canvas_draw_line(canvas, 0, y2, x2, y2);
            } else {
                canvas_draw_line(canvas, x2, y2, SCREEN_XRES, y2);
                canvas_draw_line(canvas, 0, y1, x1, y1);
            }
        } else if (abs(y1 - y2) > SCREEN_YRES / 2) {
            if (y1 > y2) {
                canvas_draw_line(canvas, x1, y1, x1, SCREEN_YRES);
                canvas_draw_line(canvas, x2, 0, x2, y2);
            } else {
                canvas_draw_line(canvas, x2, y2, x2, SCREEN_YRES);
                canvas_draw_line(canvas, x1, 0, x1, y1);
            }
        } else {
            canvas_draw_line(canvas, x1, y1, x2, y2);
        }
    }
}
```
...fixes the issue.

Have a good day!

onepointfive-REAL, 2025-08-20:
Will the developer fix it?