xMasterX/all-the-plugins #162: Enhanced sub-ghz chat not booting

Chilinwithmyflipper, 2024-10-17:
I downloaded the enchanced sub-ghz chat and my flipper just kept crashing and said "NULL pointer dereference" i know it said expect bugs it the description but I can't open open the application!

xMasterX, 2024-10-17:
your firmware version and app build? 
where you downloaded the app?
do you use external radio module or internal only?

JuanLassoVelasco, 2024-10-17:
Hey, I'm having the same issue. 
My firmware version is the recently distributed 1.0.1 official release.
I downloaded the app from the new app catalog on the flipper zero mobile app.
The app version it displays there is the 1.1 version.

JuanLassoVelasco, 2024-10-17:
I'm also using the internal radio module.

xMasterX, 2024-10-18:
I was able to reproduce the issue on the release, but it works fine on dev
There is an RC coming and next release on the way, app will not be fixed for older release since 1.0.1 had memory managment issues, wait for the next release or try Release Candidate version if its ready

xMasterX, 2024-11-16:
Official fw release is out, update and update the app in companion app, then try again, all should be fine

closed 2024-11-16 by xMasterX