xMasterX/all-the-plugins #167: [WAV Player] Regression - unresponsive control

dogtopus, 2024-11-22:
Since some recent firmware update, the buttons became unresponsive on the WAV player app. Pressing buttons, including the back button, would do nothing and eventually froze up the LCD as well.

Could be a deadlock somewhere in the input event processing routine that caused this?

xMasterX, 2024-12-06:
Can't reproduce on latest unleashed and ofw dev branch

dogtopus, 2024-12-07:
So it might be a regression on the OFW side then.

I'll test the OFW again once it gets into stable, and if I wouldn't be able to reproduce it by then, I think it should be safe to close this.

sylvathemoth, 2024-12-09:
isssue happens on momentum 008 aswell, been happening with unleashed on my flipper aswell.

dogtopus, 2025-02-13:
I can still reproduce on OFW 1.2.0.

cyber199, 2025-04-25:
Able to reproduce on official 1.3.4 as well.  A quick flick at the docs tells me I need something like the WiFi dev board to read device logs, so I can't give any more information aside from "it dun work" as of right now, unless I'm just being dense.

jakub-petrovic, 2025-04-26:
I have sane issue